garethky/Slice-Mosquito-Direct-Tool-Head #2: Tolerances for filament path should be widened a bit.

im000000, 2021-02-04:
I was having some difficulty getting my filament to feed through the opening into the hotend. I think the filament path should be widened a bit to allow for easy passage of filaments.

garethky, 2021-02-04:
I'm always looking for data to support changes like this. I have been benchmarking this design against the Prusa MK3S parts because they have so many users successfully printing and using them. I guess I never checked this particular dimension:

The filament path in the current design is set at 2.0mm. It's also has a "birds mouth" profile to combat any blockage due to bridging:
[image]

The Prusa design uses a value of 2.4mm for the filament path diameter:
[image]
see: [extruder-body.stl](https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/extruder-body.stl)

So a bit is ~0.4mm 😁, this change will be in the next commit to the dev branch. I also added the same 0.6mm filet they have around the opening. The Prusa part don't have the "birds mouth" because the split line is down the middle of the filament path, meaning there is never a bridge. The birds mouth does give slightly more room for the filament to wriggle side-to-side vs a straight tube. I'm going to remove it from the bottom, since its technically unnecessary there. If we find there is a problem with flexibles we might try removing it from the top as well.

[image]

I'll put these changes into the dev branch when I make the next update.

im000000, 2021-02-05:
thx, I will grab your dev branch over weekend.

I am now printing with PC Poly Max filament.  Instead of Prusament PC .
 My current print doesnt show any warping at least on the bed.  using an
enclosure + skirt + brim to combat this.  Also using a filament enclosure
to keep the PC dry.

thx

garethky, 2021-02-05:
Just to be very clear: I haven't pushed this change to the development branch yet! I have a bunch of changes that I am doing in the OnShape file right now:
* Soft mounting the 5015 fan
* Totally re-designed part cooling nozzle (this is my third attempt)
* Fixes to the fan shroud to prevent warping
I need to print these changes myself and validate that the tool at least passes the heater tuning test before uploading them. If all goes well I should have something to post by Monday.

garethky, 2021-02-05:
And maybe I should open all 3 of those issues as bugs here to track whats being changed...

garethky, 2021-02-08:
I just pushed this change to development along with all the other work.

im000000, 2021-02-08:
Thx , I will look at it next week and let you know how it goes.

im000000, 2021-02-09:
Hi,

I have some design improvement suggestions , specifically with hot end
block and idler --   I printed the development branch version thinking i
could replace it with main branch -- since I dont have the new fan design
--  I couldnt test it immediately but my observations still hold.

I happen to have a Prusa MK3s printer and upgraded the hot end setup with
this design:  https://mihaidesigns.com/pitstop

My main reason for moving away from Prusa's design was that I like printing
with flexibles , specifically ninja tek and it was always a hassle to
untangle the hotend every time the filament jammed.

This comes back to your design, I think the tolerances should be relaxed to
accommodate something like flexibles.  I think a flexible friendly design
will  solve other issues along the way.

I see 2 issues:
1.-- The idler design does not allow it to squish the filament, if my gears
are just a little far away, there is no corrective way to adjust for this
--prusa's mk3 design allows a lot of room to adjust the closeness of the
gears in the idler.

2.-- The PTFE tube should be more like mihai's design.
https://mihaidesigns.com/pitstop  --- the minute the gears start feeding
the filament they should direct the filament straight into the ptfe tube.
    The current design feeds filament into a tunnel that has an obstruction
which is the  ptfe tube that is embedded inside the filament path.  Think
again of flexible filament being fed into this.  If it is fed too fast it
can jam and wont go into the ptfe tube easily.

That said, I am going to take a stab at tweaking your design, if you
dont mind.    I am using Blender for my 3D work.   When I have something I
will let you know.

Thx again for your efforts.

Irumbo

garethky, 2021-02-09:
The dev branch and mainline parts are not compatible. The idler had to be shortened to make room for the 40mm fan. If you are trying to mix parts that might cause an issue.

I don't understand issue #1? Are you saying that you can't get the idler closed enough? The idler gears mesh before the face of the idler makes contact with the extruder body. The actual design has the filament squished slightly with the idler still not touching the face of the extruder body:

[image]

For #2, this design is pretty standard now. Just looking around, both the newest version of the [Prusa MK3S extruder](https://github.com/prusa3d/Original-Prusa-i3/blob/MK3S/Printed-Parts/STL/extruder-body.stl) and the [Voron Afterburner extruder](https://github.com/VoronDesign/Voron-Afterburner/blob/master/STLs/Gantry/Direct_Feed/extruder_body.stl) use this same concept. Printing the complex geometry is more reliable and provides tighter tolerances than cutting the PTFE without a jig did.

It is possible to configure parts in OnShape to have optional features, so I could have 2 versions of the extruder block, one with the feature and one with a plain hole, if you really want to cut the PTFE to shape.

garethky, 2021-02-09:
Might be worth looking at: [Filament path is not straight through Bondtech wheels](https://github.com/prusa3d/Original-Prusa-i3/issues/51)

im000000, 2021-02-09:
The idler is closing ok but design limits how close gears get together,  in
my case the filament is not being pulled through. But I can see the marks
it is making on filament.  It is also possible that the gear teeth are not
aligned properly vsomething I will verify.

I would have wanted in this design to be able to have the gears touch
together.

Prusa's design doesn't have any plastic preventing gears from getting any
closer.

I think freedom to be able to have gears touch should be a top
consideration.

In case #2 -- It was very hard for me to push the PTFE tube inside, so very
possible it didn't go in all way.

I would like to see the PTFE tube protruding a little when I open idler
door,  which will be tricky to get right because of tolerances.

I will be able to tweak the designs , have had some experiences doing that
with other designs to fit my needs.

Might use on shape as u suggest for now trying to become proficient with
blender since it is open source but may switch at some point.

It is very likely all this might be because of material I am using -- PC
but I think it is right material for this role

The again.

im000000, 2021-02-10:
I looked at the stl file for the hotend block and the measurements look
ok, with an eye to tweaking it.   Specifically the bottom part where the
ptfe tube fits.

I am going to use a smaller nozzle (2.5mm) to re-print the heater block and
see if the smaller nozzle will be a little more within spec for the
filament path.

If that doesnt work I will try and print the heater block in a vertical
position .

The main culprit I think now is the PC Max filament,  it generally prints
well and doesnt warp -- but every now and then you will see artifacts in
print where it oozes perhaps a little bit more filament than expected.

I will let you know how this goes ..

thx

garethky, 2021-02-12:
In the above linked Prusa issue they note that the "ideal" designed gap between the bondtech gears is 8.7mm. I based the gap in this design on measurements taken from a BMG extruder. I got 8.6mm. The filament path is straight relative to that spacing:
[image]

I'm going to call that "good enough".

garethky, 2021-02-12:
This is an interference projection with +2 degrees of closure on the idler beyond what is possible now. I'll see what I can do to make this work for tighter grip on flexibles.
[image]

im000000, 2021-02-12:
Don't be in a hurry to do that yet.

I re-printed the hotend block using 2.5mm nozzle and it fixed the tolerance
issue I was having with the filament path.  I also widened the ptfe tube
path that connects to mosquito by a very very tiny amount but in retrospect
i don't think that was necessary.

The result of 2.5mm nozzle  is that the ptfe tube is able to easily slide
all the way into the area created to connect it to the mosquito hot end.
This is actually good for flexibles because the ptfe tube is now occupying
the whole space and reducing  chance of flexible jamming once it gets past
the gears.

Going forward I am going to use 2.5 mm nozzle for prints that require  very
tight  tolerances.

I am now looking into why my filament is not being picked up easily, when I
start extruding.  I think this could be loose gears not being tightened.

Will know some more after some more investigation and tweaking

Once I have filament feeding without much fuss, I will have a better
opinion to make.

In general flexibles might be a tough act to solve because of the limits
you have to work with concerning  the tool changer.

Thx again.

im000000, 2021-02-14:
Hi,

I fixed the issue with one of my tools not picking filament up.   The gear
assembly that connects to the stepper motor was loose so I tightened it up.

I think you have a very good design in your hands.    However it was not
picking up  flexibles,  I think the fix for this is to allow the idler
mechanism to be able to get much closer and get a firm grip on the filament.

I am going to tweak the heater block as well to make room for the idler
gears to grind against the filament.

The design you suggested is a good start, it will also require removing
some of the material on the rotating gear assembly so it doesnt prevent the
idler gears from connecting fully.

thx

garethky, 2021-02-14:
Here you go: [Add 2 degrees of clearance to idler ](https://github.com/garethky/Slice-Mosquito-Direct-Tool-Head/commit/0b50e7e5d4a089c8dc81de069408ee751e804964 )

I don't actually think this is going to do anything for your issue though. But you can try it with the drive gear not in place and see that it closes more. Its just that the gears are fully meshed and can't close anymore. The cylinders of the Bondtech gears never touch at maximum closure (if they did we would have problems hitting the set screw!).

I tried hand feeding some TPU through an extruder (turning the reduction gear by hand so I could feel what was going on) and I didn't have a problem. You can apply a LOT of tension with this design. Its because of the leaver arm design and short spring, see [issue 6](https://github.com/garethky/Slice-Mosquito-Direct-Tool-Head/issues/6)

im000000, 2021-02-14:
Thx will try it out next after I have tested my modification.

I was also able to feed in tpu by hand with idler closed but without
stepper motor , attached and it seemed to feed in ok on original design — I
am using ninjaflex by the way which is very soft.

My guess is once the stepper motor is added it realigns everything by a
little bit.

btw Prusa doesn’t include a profile for ninja flex in their slicer filament
profile.

What I have done in my tweak is extend the length of idler by about .7 mm
around the area where the two gears meet and also remove area that would be
occupied by gear on idler  from hotend block.

If it works I will let you know.  This is what hotend block looks like (
see attachment )

im000000, 2021-02-14:
I will test your design change it is similar to what I had in mind after
looking at it.  It will take me about a week or so before I can give you an
update.   I am going to order the fans that work with this.  I might also
apply your design change to the older version also.

thx

garethky, 2021-02-15:
The idler should be all you need to print. Thank you for testing!

- Gareth

im000000, 2021-02-15:
I will print the idler today  and see how it compares to mine.

So far this revision looks to be much improved.

im000000, 2021-02-16:
I printed the idler modification you made and I am afraid to say  it cant
move gears any closer.   ( It also was a bit short in aligning properly
with the heater block in master branch.)

There is a protuberance on heater block ( see attached ), that is
preventing the idler from getting any closer.

In my modification of your design.  ( with protuberance removed )  I have
had no issues printing. I did have to tighten it progressively until I
found the right tension to grip the filament.

I havent yet tested ninjaflex filament but I am certain it will work as
well.

thx

garethky, 2021-02-16:
I don't see any pictures, can you try posting again?

im000000, 2021-02-16:
images re000 - re0001 show the modifications I made.  I lengthened idler a
bit -- not shown and cut away the plastic in front of gears.

re002 and re003 show your mostly original hotend block from master branch
with the idler from dev branch/

things to note:  I was unable to get idler in dev branch any closer to the
body of the hotend block.  so the clearance of both idlers is similar but
the one I modified is slightly lengthened.  the plastic in between both
gears when they touch together is preventing the idler from moving in
closer.

garethky, 2021-02-17:
Please only reply directly on github, email does not post images.

Here is my take on this:

This is the modified dev branch idler installed without the drive/reduction gear.  As you can see it closes until the barrel of the idler gear makes contact with the circular section of plastic that the filament feeds through.

[image]

This is the same setup, shot so we can see how closed the idler gets. Note the small gap between the idler and the extruder body:
[image]

So from the above there cant be any plastic in the way between the idler and the extruder....

This is with the drive/reduction shaft installed. I cut a hole in the back plate so we can see a little better here. Its important that the back bearing is used to check this because otherwise you will skew the shaft and the two gears won't be parallel. As you can see, the idler is not nearly as closed as it was above. The gap between the idler body and the extruder is larger. This is because the gears have fully meshed. A Bondtech gear set cant get any more closed than this.
[image]

This goes for the original idler arm as well, its no different, there was just less of a gap to the extruder body by design. And the idler does not touch the hub on the reduction gear. If it did the shaft would be the thing touching and that's not possible because Bondtech designed the gear set and the hub to have more clearance than that.

garethky, 2021-02-19:
Closing as the original request for a wider filament path has been implemented and the other issue is now moved to https://github.com/garethky/Slice-Mosquito-Direct-Tool-Head/issues/7.

closed 2021-02-19 by garethky

im000000, 2021-02-19:
the noctua fan you are testing out for new dev. branch is 12v -- is your
system 24v also?

did you follow these instructions for use of 12v noctua fan?
https://www.antonmansson.com/how-to-run-12v-fan-on-24v-3d-printer

the web page above suggests using a buck converter to bring the voltage
from 24 to 12v.    I am ordering those parts now -- It will take me perhaps
2wks or more before I test your modified design.  Want to see how well this
current one works now that the bearing is installed.  I had ordered
bearings while back but hadnt realised what there purpose was and had just
put them away.

thx again

garethky, 2021-02-20:
On the Duet 3 you can pick the voltage for both the two fan independently.

On the Duet 2 you have to pick between 12V and 24V for all fans. There are 12V 5015 fans available so you can run the Noctua. You can also get 24V 40mm fans if you want to stick with 24V, it just wont be a Noctua.

im000000, 2021-02-20:
Thx for the tip.  Didn't know this.

im000000, 2021-03-05:
Hi,

I wanted to give you an update on the toolhead design you made.

I havent yet had the chance to test the design you have in the development
branch.   Clearing up some issues I am having with a different brand of PC,
that I plan to use for printing the components.  I might stick with PolyMax
if the one I am using doesnt turn out so well.

I havent had any issues whatsoever with the older design,  been printing
mostly with ninjatek filament and so far I am not seeing the issues I used
to have on my Prusa where it might take some fiddling with the tensioning
before I could start to print with tpu.

The original issues I had with exception of the filament path were mostly
me not following the instructions properly.

So far so good and thx again.