garethky/Slice-Mosquito-Direct-Tool-Head #7: here are images showing the revised idler being obstructed.

im000000, 2021-02-17:
[
[image]
[image]
[image]
[image]
[image]

](url)

garethky, 2021-02-18:
It looks like you are missing the bearing that supports the end of the shaft and stops it from deflecting away from the filament. Without that you can totally push the drive gear back and bottom out the idler. That would explain the issue you are having:

[image]

[image]

im000000, 2021-02-18:
Thx for catching this.  It is very easy to overlook

garethky, 2021-02-18:
I will take this as feedback to include a more detailed image of this step in the instructions.