garethky/Slice-Mosquito-Direct-Tool-Head #10: dev branch fan shroud a bit too wide

im000000, 2021-04-24:
the fan shroud in dev branch that attaches to heat sink should be slimmed down so it fits the dimensions of the heat sink at point of contact.

I am using slice engineering heater cables and they are a tight fit for cable routing

garethky, 2021-04-25:
Its a tight fit but it does work. The parts thickness was actually increased to combat warping due to heat creep, see: https://github.com/garethky/Slice-Mosquito-Direct-Tool-Head/issues/5

im000000, 2021-04-27:
What material did you use for the fan shroud?  I have printed mine in PC/PBT will likely switch to PC .. I modified design to shave a few mm.  If I used PC or ABS, I will revert back to your design.  Thx

im000000, 2021-04-27:
meant to say if you are using PC or ABS then I will revert back to your design.
I had a heat creep issue on my prusa printer and it was using a mix of PC and PETG parts on an E3D V6 Hot end. I was printing using PC filament when it happened.    The PETG part that was holding the V6 Hotend was deformed but the PC parts around it did not have the same issue.

garethky, 2021-04-27:
I printed my parts form Prusament PC blend. Its one of the reasons I think I may abandon this general design. Part cooling and hot end cooling need a lot more space and the general sideways orientation of the design doesn't provide for that. Also we now have a lot of small/lite extruder choices. So the value for Tool Changer users is in the aero design which is actually the weak point of this extruder design.

im000000, 2021-04-28:
I think the design is good.  The aero is nice and small but that is it's only advantage because of the heat creep issues that has plagued it.   All about trade offs. 

I have been using the master version but will switch to dev branch but stay with stock slice fan.
The quality of the prints vs my prusa is much much better on the tool changer , I think the dev version for part cooling may be good enough.

Thx

im000000, 2021-09-13:
Hi,

I am working on an a extruder design based off of voron-0 for toolchanger ,
it is still very rough but nearly ready for testing.  I am using blender
which is not ideal for this by the way, if u want to look at it let me know.

switching to the dual 30x30 fan that the afterburner voron uses

the voron 0 project is here:
https://github.com/VoronDesign/Voron-0/tree/Voron0.1/STLs/Toolheads/Mini_Afterburner

thx